chrisys/train-departure-display #1: SPI Device Not Found

Kincade120, 2019-08-13:
Attempting to run the python code on Raspian, I get

/home/pi/.local/lib/python3.7/site-packages/luma/core/interface/serial.py:185: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/luma/core/interface/serial.py", line 280, in __init__
    self._spi.open(port, device)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./src/main.py", line 256, in <module>
    serial = spi()
  File "/home/pi/.local/lib/python3.7/site-packages/luma/core/interface/serial.py", line 284, in __init__
    raise luma.core.error.DeviceNotFoundError('SPI device not found')
luma.core.error.DeviceNotFoundError: SPI device not found

My python version is 3.7.3

 Not sure where Im going wrong here.

stuartbain, 2019-08-14:
Is SPI enabled in Raspbian?

`sudo raspi-config`

Select 5 Interfacing Options, then P4 SPI, Yes (to enable the SPI interface)

KeirDavis1, 2019-08-18:
@stuartbain That fixed it for me! Thankyou!

chrisys, 2019-08-19:
Good stuff @KeirDavis1 - thanks for helping @stuartbain. Would be great to see some photos when you finish the project.

KGS-STEAM, 2023-02-09:
Hi, I'm getting a similar error trying to run through the balena deployment, any help appreciated. I can see that the "spi=on" DT parameter is set so not sure what's going on.

Starting Train Departure Display v0.3.8
 main  /usr/src/python-packages/luma/core/interface/serial.py:206: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
 main    self._gpio.setup(pin, self._gpio.OUT)
 main  Traceback (most recent call last):
 main    File "/usr/src/python-packages/luma/core/interface/serial.py", line 309, in __init__
 main      self._spi.open(port, device)
 main  FileNotFoundError: [Errno 2] No such file or directory
 main  
 main  During handling of the above exception, another exception occurred:
 main  
 main  Traceback (most recent call last):
 main    File "src/main.py", line 325, in <module>
 main      serial1 = spi(port=1,gpio_DC=5, gpio_RST=6)
 main    File "/usr/src/python-packages/luma/core/interface/serial.py", line 320, in __init__
 main      raise luma.core.error.DeviceNotFoundError('SPI device not found')
 main  luma.core.error.DeviceNotFoundError: SPI device not found

BobBurrows48, 2023-02-10:
Having exactly the same issues as KGS-STEAM. tried on both a pi zero w and a pi zero 2 w. same outcome. Was worried I'd brought a duff screen but maybe not.

any help would obviously be really appreciated.

many thanks

Bob

gjwell, 2023-02-11:
Same here, I have tried 2 Pi zero w's two different displays. I have changed the 0Ohm resistor from R6 to R5 to enable 4SPI. triple checked wiring. metered out connections to ensure continuity and checked 3v supply to display. I guess the next thing to try is check the displays with an arduino.

any help would  be really appreciated.

Dreadmond, 2023-02-11:
Have you tried adding spi1-3cs using the “Define DT overlays” option? It’s
found under device configuration on balena cloud.

It’s mentioned in the guide for setting up dual screens, but I seemed to
need it for a single screen.

gjwell, 2023-02-11:
@Dreadmond. You are a star!!  This is exactly what I needed. Just tried it ant it works! Thank you so much for taking the time.

KGS-STEAM, 2023-02-11:
Agreed thank you so much @Dreadmond ! Worked like a charm!

chrisys, 2023-02-15:
Thanks for sharing this guys - I'll take a look here as this definitely wasn't necessary in the past!

chrisys, 2023-02-15:
Note to myself - it looks (from the logs) like the device is trying to run in dualScreen mode anyway, which explains why enabling the second SPI port resolves the error. Probably something to do with the recent changes to configuration options

jruys, 2023-02-17:
+1 - That setting fixed it for me too, thanks @Dreadmond! Already bough a replacement screen as I assumed the old one was broken.

chrisys, 2023-02-18:
OK should be fixed now - thanks for reporting it all!

closed 2023-02-18 by chrisys