chrisys/train-departure-display #3: use National rail api

joural, 2019-08-19:
I found out that transport api is very unreliable, any chance of swapping to national rail api?

chrisys, 2019-08-20:
Hey @joural what problems did you have with it? I've been using Transport API for a few weeks and never had an issue - maybe we can contact them? 

But otherwise I've no objections to adding the National Rail API as an option for people, it's good to have more options, right? 👍

joural, 2019-08-20:
The information from transport API doesn't seem to be as accurate as from national rail, national rail gets information directly from all TOCs. Like yesterday my screen was showing service as running late, when in reality it was starting late at start station, and was not longer calling at my stop=was cancelled, transport API didn't pick this up, same when there was distribution on my line, it was showing services that were meant to leave hour ago, as running on time...

joural, 2019-08-20:
I'm trying to change your code to take national rail, but is proving very difficult for my novice skills.
One thing to note is national rail API requires to display disclaimer message periodically and uses soap

madwort, 2019-08-23:
Hi @joural, 

Sorry to hear you've seen some poor data quality on TransportAPI. We get our train data from amalgamating several different open data feeds direct from Network Rail, which allows us to make the data available for your use with no restrictions (disclaimer messages, rules on what the data can be used for, etc), and in a hopefully very convenient form (JSON API, rather than SOAP). We do sometimes diverge slightly from the National Rail Enquiries feed (for instance, the TOCs like to make assumptions about how much time late trains will make up between stops - which is not open data - so we give slightly different results in that case), but we're always trying to improve our data quality.

If you see specific instances where we're diverging significantly, please email us at support@transportapi.com with endpoint URL, observed API output & expected API output, and we'll try to look into what's happening. It's not always possible to look back in time with live train data, but it can tell us where to look to make improvements in future.

Thanks! 
Tom Ward, TransportAPI

joural, 2019-08-23:
@madwort I appreciate the quality of data supplied by transport API, without being connected to the source (national rail), I can only imagine the lenghts to gather all that information. I will keep my eyes peeled and report 😉

madwort, 2019-09-13:
Hi @jajsilver ,
Yes, I can see that would be useful, I'll discuss with colleagues & file a feature request. 
Bear in mind that if you're setting up a customised departure board for your own use, you can filter the live departures like this: 
https://transportapi.com/v3/uk/train/station/STP/live.json?destination=SUO&calling_at=WIM
(shows only Sutton via Wimbledon trains)
Or alternatively, I think you can do further requests to TransportAPI's `/train/service` endpoint with the `train_uid`s from the live departure board to check their routes (but obviously that's going to generate a lot more requests...!)

jajsilver, 2019-10-06:
@joural @chrisys I set myself this as a project and have now had a go at rewriting to use National Rail OpenLDBWS (Darwin feed) in an attempt to teach myself Python! See my fork here:
https://github.com/jajsilver/UK-Train-Departure-Display-NRE
It now accommodates multiple destinations (trains that split) and the via attribute. 
If you get the chance to test it out and let me know how it goes for you I'd be very grateful.

chrisys, 2019-10-08:
@jajsilver awesome, great work! I will give it a go sometime this week hopefully.

joural, 2019-10-08:
@jajsilver truly awesome, I will as well give it a go when I ll have spare time, thank you it truly is appreciated

closed 2019-10-08 by joural