chrisys/train-departure-display #9: departures environment variable is not set

rogerw-gh, 2019-09-16:
Worked through the configuration - which is brilliant BTW, I've already got Chris Hutchinsons' version working.  The balena console logs in balena cloud is constantly going round complaining about the 'departures' environment variable not being set.  I've tried setting one and restarting the Pi and its still complaining..  Have set departures as an environment variable in the Application Environment variables and in Device Service variables - I know the hardware is good - all I've done is swapped the microSD and deployed the application and watched it deploy to the device.

chrisys, 2019-09-19:
@rogerw-gh can you post screenshots of your configuration variables and the log output? Just asking as the variable isn't actually called `departures`, it's `departureStation` as per the [configuration section](https://github.com/balena-io-playground/UK-Train-Departure-Display#configuration) in the readme.

chrisys, 2019-10-08:
@rogerw-gh closing this for now but feel free to re-open if it's still an issue when you get a chance to look at it.

closed 2019-10-08 by chrisys