chrisys/train-departure-display #11: Can't seem to set timezone

Chuckatron, 2019-10-20:
Hey - I finally got round to building this and it's working great, so thanks very much for your work on this! I'm trying to change the timezone from the default which appears to be UTC. I've read around and it seems I should have a `balena-timezone` program on my host OS, but I don't appear to:
[image]
I'm sure I must be missing something simple! Any idea how I can go about changing the timezone, thanks!

chrisys, 2019-10-21:
Hey @Chuckatron have you tried setting the `TZ` environment variable for the device? If you set it to the 'TZ database name' as per this list: https://en.wikipedia.org/wiki/List_of_tz_database_time_zones

That should work - let me know if not!

Chuckatron, 2019-10-21:
Hi @chrisys Yes, that's how I have it set, at least I think it's right:

[image]

When I check the host's timezone, it seems to think it's UTC:
```
root@dee6f19:~# date
Mon Oct 21 16:14:16 UTC 2019
```
Container is slightly different:
```
root@dee6f19:/usr/app# date
Mon Oct 21 16:14:50 Europe 2019
```
Any ideas on anything I might have got wrong?

chrisys, 2019-10-21:
@Chuckatron interesting, what balenaOS version are you using?

Chuckatron, 2019-10-21:
@chrisys I'm using balenaOS 2.32.0+rev1

chrisys, 2020-05-26:
This seems to be fixed. @Chuckatron please do repoen this if you're continuing to have issues with it, I'd also suggest trying a fresh build in order to get the latest base images.

closed 2020-05-26 by chrisys