chrisys/train-departure-display #18: `config.json` not pushed because it is listed in `.gitignore`

pdcastro, 2020-04-10:
There is logic in the [balena-run.sh](https://github.com/balenalabs/uk-train-departure-display/blob/master/balena-run.sh) file to decide whether to use an existing `config.json` file or copy from `config.sample.json`.

Having seen that logic, I [advised a user in a forum thread](https://forums.balena.io/t/local-mode-not-working/92301/25) to create a `config.json` as I thought it would be more convenient than setting the environment variables on the command line when pushing to a local-mode device:
`balena push <ipAddress> --env var1=value1 --env var2=value2 ...`

The tried it and it failed. On investigation, I found that there was a line for `config.json` in the repo's `.gitignore` file. The CLI (`balena push`) currently considers both the `.gitignore` and `.dockerignore` files to decide what files to ignore (balena-io/balena-cli/issues/1032) when pushing to the cloud or to local devices, so the `config.json` file was not being pushed. Deleting the line for `config.json` in the `.gitignore` file solved the issue.

(If it is a design decision that `config.json` should not be pushed, then what would the use case be for the test in `balena-run.sh`?)

pdcastro, 2020-04-11:
The project's `config.json` file contains an API key (`transportApi.apiKey`) and it is not good practice to commit keys to git repositories, especially of course public ones. But removing the `config.json` line from the `.gitignore` file does require committing `config.json` to the repo: one could live with `git status` reporting "untracked files" for some time. Progress is being made on CLI issue balena-io/balena-cli/issues/1032, which will result in the CLI no longer considering the `.gitignore` file for what to push (looking at the `.dockerignore` file only). Once this CLI issue is resolved, the `config.json` line could be re-added to the `.gitignore` file.

chrisys, 2020-04-11:
@pdcastro you got it, the `config.json` was added to `.gitignore` for the reason of not committing these private keys. I've always worked with setting the variables on the command line when developing in local mode for this project.

pdcastro, 2020-04-11:
> I've always worked with setting the variables on the command line when developing in local mode for this project.

Thanks @chrisys. I gather that using a `config.json` file would similarly not work when pushing to the cloud, whether `git push` or `balena push`. So basically, at the moment (with `config.json` listed in the `.gitignore` file), it is not possible to use a `config.json` file with this project, whichever the push method or target.

The logic in the `balena-run.sh` script is that, if a `config.json` exists (currently "impossible"), it is used as is. If it doesn't exist, a `config.json` file is generated with the values from environment variables (whether the variables are set on the command line, or via the web dashboard). `balena-run.sh` runs on the device, not the development machine, so an auto-generated `config.json` file will exist on the device, not on the development machine.

I think it makes sense to remove `config.json` from `.gitignore`, which is not the same as requiring or recommending that users commit the file to some git repo (`balena push` does not require it). It is not even necessary to document the possibility of writing a `config.json` file. I am not proposing any changes to the Readme / documentation.

Then again, another option is to do nothing (in relation to this issue) and wait for CLI issue balena-io/balena-cli/issues/1032 to be resolved, at which point `balena push` will allow `config.json` to be pushed even if it is listed in the `.gitignore` file.

PR #19 (2020-05-26): Improve local development
Fixes #18 

Add a config.json which is used for default variables (meaning they are not required to be specified on commandline).
Make balena-run.sh override config.json with any environment variables that *are* set
Add use of PiWheels making local builds on Pi Zero much much faster (a few minutes for a clean build)

Change-type: patch
closed 2020-05-26 by bulldozer-balena[bot]