chrisys/train-departure-display #26: Tweak to add London Underground stations

scotta228, 2020-12-03:
Would it be possible to add option via new variable and amendment in trains.py to allow the alternative web call?
London underground data calls require the Line name to be included in the call.

New variable: SERVICE_TYPE - value is text:    null / blank / train / tube        (I haven't looked at other networks)
New variable: SUB-TYPE - value is text: null / blank or text string [A-Z|a-z]*

URL - with null/blank entry in SERVICE_TYPE (presently in trains.py at line 22):
URL = f"http://transportapi.com/v3/uk/train/station/{departureStation}/live.json"

Non-null: 
URL = f"http://transportapi.com/v3/uk/(service_type)/(sub_type)/(departureStation)/live.json"

e.g. 
url = f"http://transportapi.com/v3/uk/tube/metropolitan/CLF/live.json"
url = f"http://transportapi.com/v3/uk/train/station/CFO/live.json"

of course these could be extended with another optional variable: PLATFORM - value null/blank or integer

if null / blank - url as above
not null/blank 

URL = f"http://transportapi.com/v3/uk/(servicetype)/(line)/(departureStation)/(platform)/live.json"

Summary of suggested variables:

SERVICE_TYPE   "train" for Netwok Rail or "tube" for London Underground
SUB_TYPE          "station" for Network Rail or (underground line name) for London Underground
departureStation - no change
PLATFORM       (optional)

chrisys, 2021-01-18:
@scotta228 I'd love to see this implemented, it'd be great to get this into the project as an option.

scotta228, 2021-01-18:
Actually, there is a different project on GitHub using the TfL OpenAPI which better suits the tube. TfL data is based on arrivals, rather than departures, and there are problems at the end of lines (where the route starts), so it's far from perfect. I've tweaked this other project to work with the Balena architecture, but it needs to remain a completely separate project, as it uses a different data source with a completely different data hierarchy.

closed 2021-01-18 by scotta228

chrisys, 2021-01-19:
@scotta228 what project is that?

scotta228, 2021-01-19:
https://github.com/petykowski/London-TFL-Arrivals-Board