chrisys/train-departure-display #29: error message - needs clearer information

satellitejack, 2021-01-27:
I received an error when using Balena desktop of:
Error: Please ensure the ‘departures’ environment variable is set

well there is no global variable called 'departures' cane you please rename to the actual variable required

daneverest, 2021-03-14:
I'm having this problem as well. How did you fix it?

stevecady, 2021-07-06:
I'm having the same issue too - did you manage to resolve it?

satellitejack, 2021-07-06:
Work on it likethis: it does work so what am I doing wrong?
Memory, dont skimp, get a good one, like Sandisk
The wifi only works at 2.4GHz

Use your phone as a hotspot, that works too.

closed 2021-07-08 by chrisys

chrisys, 2021-07-08:
@satellitejack I think this is addressed in the latest version of the code now - please feel free to reopen and let me know if not!

commit 7bc059a1 (2021-07-08): Report API server errors fixes #29, fixes #32

Resolve the case where if an API key/App ID is not specified it causes the code to report that the 'departures' variable is not set. This really means that it could not find the departure data in the API response because it was 500 erroring.

Change-type: patch