chrisys/train-departure-display #32: Error: Please ensure the 'departures' environment variable is set

stevebonito, 2021-05-31:
Great walk through and deployment info, struggling to set live and getting 'Error: Please ensure the 'departures' environment variable is set' message in Balena Log, just keeps looping round leading to no output to the oled screen.

Error code below:
```
main  Error: Please ensure the 'departures' environment variable is set
(date time etc) Service exited 'main sha256:da3734455afb949d9d9a7535470d0377c7722abec1eac95a03929e7fd6fc9a2f'
```

stevecady, 2021-07-06:
I'm having the same issue - did you manage to resolve it?

closed 2021-07-08 by chrisys

chrisys, 2021-07-08:
@stevebonito this should be addressed in the latest version of the code now - please feel free to reopen this if not!

commit 7bc059a1 (2021-07-08): Report API server errors fixes #29, fixes #32

Resolve the case where if an API key/App ID is not specified it causes the code to report that the 'departures' variable is not set. This really means that it could not find the departure data in the API response because it was 500 erroring.

Change-type: patch