chrisys/train-departure-display #46: Raspberry Pi Zero WH maxes out at 100% constantly

mattanorak, 2021-11-04:
Looking in the dashboard, I see that my RPi Zero WH is constantly maxing out at 99-100% CPU and the effective FPS is anywhere from about 2.2 to 2.22 - consequently the refresh on the screen (when text scrolls) is very slow and jumpy rather than smooth as with the Pi3.  I also notice that the clock at the bottom sometimes misses a second or catches up after a pause (especially at the time when scrolling begins on the 'calling at' stations).

It could well be that the Pi Zero WH is just not powerful enough for this, but wanted to raise it in case it's something else causing an issue with this device.

This is only for the Pi Zero WH - the Pi3 is fine.

chrisys, 2021-11-05:
This is something that's bugged me for a long time, and I have had a few different debugging sessions with no real improvement. I did the original development work on a Pi 3 and like you found it to be a lot smoother, but had to sacrifice the performance for the convenient form factor of the Zero. However with the recent release of the Zero 2 I am hoping we can have the best of both worlds! I have one arriving in a few days and will put it to work in my sign when I get a chance.

mattanorak, 2021-11-06:
I've got one coming next week as well.  What device would it be under in Balena Cloud?

LeePorte, 2021-11-07:
I have a zero 2 and am seeing 78 - 82% CPU utilisation.

chrisys, 2021-11-09:
We're still working on adding the device type to balenaCloud but it shouldn't be too much longer. This is the issue to follow for updates: https://github.com/balena-os/balena-raspberrypi/pull/725

chrisys, 2021-11-09:
@LeePorte what effective frame rate does the code report in the logs when you're running on the Zero 2?

LeePorte, 2021-11-09:
I've seen highs of 17.91 and lows of 10.86

```
Nov 08 11:14:52 train-sign python3[2778]: Effective FPS: 10.86
Nov 08 11:14:52 train-sign python3[2778]: Effective FPS: 10.88
```

```
Nov 09 08:48:29 train-sign python3[10163]: Effective FPS: 17.9
Nov 09 08:48:29 train-sign python3[10163]: Effective FPS: 17.91
```

mattanorak, 2021-11-10:
Excellent - might give that a try while I'm waiting for the device type to appear on BalenaCloud (physical device arrived today).

LeePorte, 2021-11-10:
If it would help I can post up the systemd service file.

mattanorak, 2021-11-11:
Exellent - if you can!

LeePorte, 2021-11-11:
Here's my service file to use as a base

```
[Unit]
Description=Train Sign Service
After=multi-user.target

StartLimitIntervalSec=500
StartLimitBurst=5

[Service]
Type=idle
WorkingDirectory=/home/pi/uk-train-departure-display
Restart=on-failure
RestartSec=5s
Environment="TZ=Europe/London"
Environment="departureStation=<STATION_CODE>"
Environment="refreshTime=120"
Environment="screenRotation=2"
Environment="apiKey=<API_KEY>"
Environment="operatingHours=8-22"
Environment="screenBlankHours=1-6"
Environment="outOfHoursName=<STATION_NAME>"
ExecStart=/usr/bin/python3 src/main.py

[Install]
WantedBy=multi-user.target

```

mattanorak, 2021-11-11:
Perfect - thank you.  I'll give it a go later tonight on the Zero2 and let you know how I get on.

mattanorak, 2021-11-14:
Thanks, @LeePorte - For some reason I couldn't get the service file to work, so instead I edited config.py (I also had to put VERSION in the home directory (I think that was a Path statement thing though - or lack thereof).
What I did was just create a cron entry to run on boot and that did the trick.
Before I could get it to work, I had to do a few things like install the dependencies in the requirements.txt file, but that was easy!
Thanks for the tip!
Currently getting about 19.88 FPS standalone on the Zero2W.

mattanorak, 2021-11-15:
@LeePorte - Odd, I've noticed that this seems to stop running after a while (after 30 mins this morning).  I checked using:
`ps-ef |grep main`
and I didn't see the main.py process running as I normally do, so this seems to have terminated.
I am running the command with:
`python3 /home/pi/uk-train-departure-display/src/main.py &`
It's running in the cron as an @reboot, so rebooting fixes it - temporarily.

Anyt thoughts on this?

LeePorte, 2021-11-15:
I had seen some network timeouts that caused it to die. Though since using the systemd service file, I've not had that issue as it has restart on failure set.

I used the following to create and enable the systemd service file

```
sudo vi /lib/systemd/system/train-sign.service 
sudo systemctl daemon-reload 
sudo service train-sign start
```

What method did you use to try and install it?

mattanorak, 2021-11-15:
I seem to remember doing something similar (with nano, though, not vi).
what did you put in `train-sign.service`?

LeePorte, 2021-11-15:
Nano is fine, use your editor of choice

The contents of train-sign.service was the following with some of the details adjusted (<STATION_CODE>, <API_KEY> and <STATION_NAME>) to suit my info.

```
[Unit]
Description=Train Sign Service
After=multi-user.target

StartLimitIntervalSec=500
StartLimitBurst=5

[Service]
Type=idle
WorkingDirectory=/home/pi/uk-train-departure-display
Restart=on-failure
RestartSec=5s
Environment="TZ=Europe/London"
Environment="departureStation=<STATION_CODE>"
Environment="refreshTime=120"
Environment="screenRotation=2"
Environment="apiKey=<API_KEY>"
Environment="operatingHours=8-22"
Environment="screenBlankHours=1-6"
Environment="outOfHoursName=<STATION_NAME>"
ExecStart=/usr/bin/python3 src/main.py

[Install]
WantedBy=multi-user.target
```

It might we worth checking your python3 path with `which python3` and making sure it matches.

mattanorak, 2021-11-15:
Yep, I set python3 to run as default and in the script anyway, I call python3 ....
I also found that running it like this will also restart the process (I think) if it terminates:
`setsid python3 /home/pi/uk-departure-display/src/main.py < /dev/zero &> /home/pi/log.txt &`

Next, if this keeps terminating (which it seems to be OK up to now), I'll try belt and braces like this:

`!/bin/bash
if [ $(ps aux | grep 'yourscript.py' | grep -v grep | wc -l | tr -s "\n") -eq 0 ]; then /your/path/to/yourscript.py; fi &`

Messy, but it looks like it would work :-)

mattanorak, 2021-11-15:
OK, I got it working as a service now.  I'll see how long it stays up.  The only difference I used here (and that seemed to get it working) was instead of:
`After=multi-user.target`
I used:
`Requires=network.target`
and then in the main section I added:
`User=pi`

Fingers crossed....

LeePorte, 2021-11-15:
@mattanorak How's it been holding up?

LeePorte, 2021-11-15:
Interestingly CPU usage seems to be a little lower under Bullseye

[image]

mattanorak, 2021-11-16:
Seems to be OK - stayed up all yesterday and is still there this morning :-)
CPU seems roughly the same as yours.
[image]

Thanks for the assistance - will keep an eye on it and see how it goes.

chrisys, 2021-12-06:
Hey @LeePorte @mattanorak since balena now has support for the Pi Zero 2 I wonder if you guys have given it a try?

LeePorte, 2021-12-06:
I confess i have not. I've been working on the dual display. If I get a chance I'll give it a spin in the next few days.

mattanorak, 2021-12-07:
Hi - also not had chance to test.  I'll see if I can do that this week.  I've got a couple of Zero W 2's now and some spare memory cards, so if I get some time I'll give it a whirl.

chrisys, 2021-12-09:
Just a heads up guys, I haven't attached a display to my Zero 2 yet but I did try building the project for the Zero 2 and it fails. The way to get it to work at the moment is to set the default device type for the balenaCloud application to the Zero 1 (as you would have done before), then add a Zero 2 to the Zero 1 application. This way the builders build for the 32 bit architecture of the Zero 1 but it still runs on the Zero 2 (64 bit) as it's backward compatible. I will look into the changes necessary to update the project to make it build for 64 bit architectures.

chrisys, 2021-12-10:
Scratch that, it should work now due to the workaround in #56

LeePorte, 2023-07-07:
@chrisys I've just updated and it's running much faster. Possibly a bit too fast. Though that may just be me being used to how it was before. I'll try and grab a video.

mattanorak, 2023-07-08:
Must admit I've not tried it in the dashboard for a while, I've been running standalone - I'll plug my spare in and give it a go.

LeePorte, 2023-07-08:
Here's a [video of it running on a PI zero 2W](https://cloud.porte.uk/index.php/s/GSfhsPdpYqVnsci)

And the resulting htop

[image]

Definitely much lower resource use

chrisys, 2023-07-11:
@LeePorte oh wow that's definitely scrolling too fast for me! We might need to close this issue and start a new one for scrolling speed to fast now - whodathinkit!

LeePorte, 2023-07-11:
It's much easier to slow things down though!