chrisys/train-departure-display #47: Multiple displays

LeePorte, 2021-11-07:
Hi,

What do you think about the option to have multiple displays? The Pi could support this as it has 2 i2c busses (https://github.com/rm-hull/luma.core/issues/55#issuecomment-286868008).

My use case is that I have a 2 platform station and would like to have one display for platform 1 and one display for platform 2.

LeePorte, 2021-11-23:
I've got it working on my fork

[image]

In the case of there being no platform assigned it is skipped in the filter until the train is assigned a platform.

Would there be any interest in raising a PR for this? As it is controlled by environment variables the single display behaviour is not affected.

chrisys, 2021-12-06:
@LeePorte absolutely! This looks really cool. I'm interested in seeing your implementation but otherwise have no objections to this feature being added. Again my apologies for the late reply as I've been offline for a couple of weeks.

LeePorte, 2021-12-06:
No worries about the time, real life is a thing we all have!

I've created https://github.com/chrisys/uk-train-departure-display/pull/54 for you to have a look at for my implementation. There is probably plenty of optimisation that can be done.

LeePorte, 2021-12-29:
All merged in so closing the issue. Thanks for taking the time to review the issue and the PR

closed 2021-12-29 by LeePorte