chrisys/train-departure-display #50: GPIO Issue - Channel already in use

AtomicLaser, 2021-11-16:
Hey all, having got this working, I am now having an issue with a GPIO warning - Logs below if anyone knows how this can be resolved?

I am running on a Raspberry Pi3 

main  Starting Train Departure Display v0.1.1
 main  /usr/src/python-packages/luma/core/interface/serial.py:200: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
 main    self._gpio.setup(pin, self._gpio.OUT)

mattanorak, 2021-11-19:
I think it's quite normal if the program has already been running and you're running it again.  As far as I recall, unless it doesn't start, you can ignore it, but I'm prepared to stand corrected.

chrisys, 2021-12-06:
What @mattanorak says is my understanding too; we can probably handle this in a more graceful manner but as it stands I've seen the message a lot but it's never actually caused an issue. I'll leave this issue open as a reminder for us to do that :)

AtomicLaser, 2021-12-06:
FWIW, my original issue with this is that the screen blanked at the same time as the log message and would not resume.

After doing some investigation, I think there was an issue with the solder blob on the 4SPI jumper, so after fixing that I've not had any other issues, although as noted, it still does show up in the logs from time to time without any further issue.