chrisys/train-departure-display #65: Documentation: API provider changed

jruys, 2022-02-17:
You may want to add a note in the documentation that the API provider seems to have changed from TransportAPI to OpenLDBWS API. If you have a setup where you have a separate Application ID and Application Key - you need to register for the new API.

Had me scratching my head for a bit to work out why my setup had stopped working.

chrisys, 2022-02-17:
@jruys thanks for the prompt - will do so!

Did you manage to get the project running OK again with the new API provider?

jruys, 2022-02-17:
> Did you manage to get the project running OK again with the new API provider?
 
Yes, once I realised I had to register for a new API key and fill it in the apiKey field, it all started working again! What triggered me is that the key in the documentation looked different from mine and the Application ID had gone 😄

commit 0ef0e04d (2022-02-23): Document API provider change fixes #65
closed 2022-02-23 by chrisys