chrisys/train-departure-display #66: Display turns blank after a few days - needs a reboot

mattanorak, 2022-02-23:
Sorry I can't be a bit more specific here.  I've just come back and noticed that the displays on both my Pis are blank - this seems to happen after a few days.  Only a reboot brings them back on.

I've not updated the APIKey to the new format - but I think this was happening before that.  I suspect they can't get a connection or something at some point and then just time out and never bother trying to connect again.

I'm running standalone, not in Balena Cloud.

Any initial thoughts?

chrisys, 2022-03-01:
@mattanorak I'm not really sure, I'd suggest updating to the latest version anyway as there are other changes and fixes after the API was changed that may help. If it's still happening once you're running on the latest version then we might stand a better chance of figuring it out.

IIRC you were using the Pi Zero 2 right? I have since updated my unit to the same device and am running it on balenaCloud with the latest version and haven't encountered any issues in quite some time now - touch wood!

mattanorak, 2022-03-04:
OK, thanks, Chris.
Out of interest - the OpenLDBWS API Key referenced in the instructions, do we use that one (ie is it the same for everyone, or is that just an example: f1ff949c-769e-49b3-8242-eee54fe91b94)

Matt

closed 2022-05-09 by chrisys