chrisys/train-departure-display #67: Departure board on steroids....

Stavrosi, 2022-03-02:
Hey Chris, hope you're keeping well chap.

I stumbled across your awesome Pi train departure board project after a random conversation with a mate over the weekend. It got me thinking that it would be very cool if I could, for his birthday, make him a nice display to go on the wall of his pub to display the train times (his pub is right next to our local station). However, to make this useful, it would need to have a much bigger screen attached to it (something like an 11.9" HDMI connected panel for instance). My fear here is that simply outputting the data to an "normal" screen would mean it would lose all of it's authenticity therefore, a larger matrix style screen would really make it pop!

I've seen (in the comments) someone else asking about this however, with my very limited knowledge, I'd have no chance of getting it working with a different type of display. To this end, I was just wondering if you were aware of anyone who may have already tackled this or, perhaps there's someone you know that could maybe take a look at it?

Kind regards,

Steve.

chrisys, 2022-03-02:
Hey Steve, that sounds like an awesome project!

What's your experience like with assembling electronics? Could we use something like this? https://www.amazon.com/Indoor-128x64-Module-256mm128mm-256128mm/dp/B0869P1DCH

I think it would be possible to use 2 of the above to create the same 256x64 matrix as the small OLED screen, and make something pretty awesome to go in the pub. The above would give you a 500mm wide display. If you're up for it I'd be willing to figure out what would be the right panels to get and make sure the software works with them, but of course the assembly and hardware side of things would be up to you. Honestly there may even be a cheaper version as that one is full RGB and we obviously only need one colour.

Using HDMI monitors is possible - you'll note there's an open PR in the repo on GitHub which adds this, but I don't think it looks anywhere near as impactful, and I've been resistant to merge the changes into the project as it adds a lot of bloat to the software.

What I would ask though is that we move the discussion to an issue on the GitHub repo so everything is in the open and other users of the project can see and contribute as well though.

Let me know your thoughts.

chrisys, 2022-03-03:
I ended up ordering a couple of those displays, I'll give it a try when they turn up and see how it works out 👍

Stavrosi, 2022-03-03:
> CAUTION: This email originated from outside of the organisation. DO NOT CLICK links or open attachments unless you recognise the sender and know the content is safe.

Morning Chris, thanks for the update.

I’m inclined to order a pair also so that I can get to work on an enclosure to suit the panels as soon as they arrive.

Thanks again,

Steve.

chrisys, 2022-03-03:
Up to you Steve but no guarantees we can get it working at this stage, I mean to a certain extent it's just software so I am sure we can figure something out, but you may want to hold off.

chrisys, 2022-03-23:
[image]

Hey @Stavrosi I got a proof of concept going! Will work on polishing it up and making relevant PRs. 👍

GOTO-GOSUB, 2022-03-23:
Now that is looking good.

chrisys, 2022-03-23:
It looks so good! There's something oddly satisfying about miniaturising something then making a bigger version of it again 😂 

It's all relatively simple stuff, the main piece of work is adding support for these displays to the luma.led_matrix library (for which I'm creating a PR), then some minor tweaks to this project to utilise that. 

A note for @Stavrosi though, you'll probably need a board like this: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

I've connected the displays directly to the GPIOs on a Pi 3 for testing but get some flickering. I am assuming this will be resolved by one of the above boards, so I have ordered boards and parts to make a few. If they all work OK I'll happily offer any extras to people building this project for parts + shipping costs.

chrisys, 2022-04-06:
@Stavrosi my PCBs and parts for the active-3 board are due to arrive within the next few days - do you want me to assemble an extra board and ship it to you for testing?

chrisys, 2022-04-18:
PCBs arrived at last! Still waiting on components.

[image]

chrisys, 2022-05-18:
At last my component orders have arrived so will put together some boards and keep testing. Do you need me to send you a completed board if it works OK @Stavrosi ? I have 5 PCBs so much more than I need :)

mattanorak, 2022-06-13:
Hi @chrisys - Just wondering how you're getting on with this.  I'd be interested in a kit list and instructions for putting one together if you've got it working.
Is that a colour LCD board?  If so, are you planning on multi colour or just the one colour?

chrisys, 2022-07-01:
@mattanorak sorry I've not had a chance to push this forward any more yet, but yes it is an RGB LED matrix panel, so in my proof of concept had coded it so you can specify the colour of the display with an environment variable (single colour).

Stavrosi, 2023-07-27:
Wow, it feels like so long ago that we first started chatting about all of this. I had to go away with work for a few months and completely lost the thread on this when I returned. I'd still very much like to get the large format up and running - it seems that stock of Pi Zero 2W's is now levelling out a little so just ordered one too.

mattanorak, 2023-08-23:
I’ve got a couple of spare PIs in and I’ve also got 3x P5 full colour LED boards and connectors.  Happy to test along - what else is needed?MattMatt Newton

christopherpm, 2023-10-03:
If the large version works, please post details - I'd love to build it!

chrisys, 2024-04-01:
Sorry guys some stuff came up; I still have all the hardware for this and do intend to get back to it one day!

If anyone wants to hack on it though all you need is here: https://github.com/rm-hull/luma.led_matrix/pull/275
And here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md