chrisys/train-departure-display #69: Option for departures from a specific platform

mattanorak, 2022-04-23:
I was sure there was an option for this once, but I can't see it in the variables now.  Could you please add the option to be able to filter by only specific platforms - eg Platform 2.  You can do this on multi-screen, but can't see a way on single screen.
Thanks.

mattanorak, 2022-04-23:
Ah, I figured it.  Worth documenting though, for those running single screen, just add:
`Environment="screen1Platform=2"`
(or whichever platform you want to single out).

chrisys, 2022-04-29:
Thanks @mattanorak this is mentioned in the table of configuration variables in the readme, just wondering if what's there isn't clear enough or that you didn't originally find it?

mattanorak, 2022-05-02:
Hi @chrisys ,
In the configuration table, I initially read this to mean that the 2 variables were only valid if you had 2 screens set up (so if you have `dualScreen=true`.
Maybe an additional comment in the table of `screen1Platfom`, to say that this also works for a single screen environment (without having to set `dualScreen=true` if you want to display a single platform on a single screen setup.

commit 8caffd5f (2022-05-09): Readme updates, also fixes #69

Change-type: patch
commit 59880495 (2022-05-09): Merge pull request #71 from chrisys/update-readme

Readme updates, also fixes #69
closed 2022-05-09 by chrisys