chrisys/train-departure-display #72: Acceptable for variables?

ScottTheHippy, 2022-05-16:
Howdy- I'm trying to get the screen to show the time 24/7.  What I'm finding is that the variable(s) operatingHours, screenBlankHours,  and/or outOfHoursName seem to be in effect even if deleted.  I tried setting screenBlankHours to "0" and that didn't seem to make a difference.  Is there some combination of these variable settings that will lead to the clock showing 24/7?   Thanks, and fun thing to build.  I had the same screen as you, and I *might* have used a lot of bad language moving that resistor.  :)

ScottTheHippy, 2022-05-17:
As a follow up, it seems the big hurdle is screenBlankHours.  If I delete the variable, it operates on the 1-6 default.  What disables this all together?  I'll try adding it back and setting it to 0-0.

chrisys, 2022-05-17:
@ScottTheHippy I've just written a fix for it to allow you to run with blank variables, when #73 merges give it a try and let me know!

chrisys, 2022-05-17:
P.S. thanks for building the project and thanks for reporting this! I am glad you had fun :) Feel free to reopen this issue if this does not fix it.

commit 4e6790fd (2022-05-17): Default to all-day operation fixes #72

Change-type: patch
commit 6b83839e (2022-05-17): Default to all-day operation fixes #72

Change-type: patch
commit 07b20e76 (2022-05-17): Default to all-day operation fixes #72

Change-type: patch
commit 277fce9a (2022-05-17): Merge pull request #73 from chrisys/patch

Default to all-day operation fixes #72
closed 2022-05-17 by chrisys