chrisys/train-departure-display #78: If set to station WBQ, screen is often blank, even after reboot.

mattanorak, 2022-09-22:
Odd one to track down the issue, but if I have the board (running locally) set to WBQ as the station, it can work fine for a while, but then the screen will go blank.  It will appear to cycle through the start screen a couple of times and then remain blank.
Rebooting the Pi goes through the same cycle.
If I change the station to anything else, it seems fine.

Not sure where to look for logs to see why nothing is showing - any tips appreciated.

chrisys, 2022-10-24:
@mattanorak you should be able to use the logs output in the balenaCloud dashboard - for example in #77 the logs were shared (which I have a hunch is probably related).

mattanorak, 2022-10-24:
I'll give that a try - thanks.  I suspect it's the Pi Zero's Wi-Fi going to sleep and not recovering properly, but will try this.

Thanks.