chrisys/train-departure-display #92: Unable to use due to API restrictions

davebradford, 2023-03-17:
Hello!
I’ve enjoyed building this project a few years ago and found it again so decided to fire it up.

I notice that there has been some changes to the free tier of the API limiting to 30 requests per day.

I wondered if anyone else has encountered this and if there was a way I could restrict the project so that it only checks 30 times per day? 

Many thanks

/d

jruys, 2023-03-17:
The project switched API provider a while back for this very reason (see issue 39). I’m guessing you may still be running the same software from a few years ago which runs into this limit.

Re-deploy the software and follow installation instructions again (you’ll have to get a different API key) and you’re probably back in business.

chrisys, 2023-03-21:
+1 to what @jruys mentioned (thanks for helping out!). The instructions/documentation on the repo are a lot better these days too so hopefully not too much work to update.

Give this a try and feel free to re-open the issue if you still have problems.

closed 2023-03-21 by chrisys