chrisys/train-departure-display #93: Getting 3 dots on the left of the display and no departures are showing.

ISC82, 2023-04-26:
So I did this build a few weeks ago and it has been working flawlessly. I am happy with it I am thinking of getting myself a 3D printer so I can create a few and gift to friends and family. However, today it started showing 3 dots on the left of the screen and no departures are being displayed. It is still showing the "Welcome to [My local station]" and the time is correct and updating. Diagnostics is showing no errors. I tried lots of other station codes with the same result on each. I even re-imaged the build to a different SD card but get the same result. This has been for at least the past 12 hours. Any suggestions? 

I have posted a screenshot on my Twitter.

https://pbs.twimg.com/media/FurM7sAWYAADxKd?format=jpg&name=small

ISC82, 2023-04-26:
Here are the logs after a reboot. 

26.04.23 22:57:52 (+0000)  main  Starting Train Departure Display v0.4.0
26.04.23 22:57:52 (+0000)  main  /usr/src/python-packages/luma/core/interface/serial.py:206: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
26.04.23 22:57:52 (+0000)  main    self._gpio.setup(pin, self._gpio.OUT)
26.04.23 22:57:58 (+0000)  main  Effective FPS: 0.0
26.04.23 22:59:59 (+0000)  main  Effective FPS: 11.74
26.04.23 23:02:00 (+0000)  main  Effective FPS: 11.68
26.04.23 23:04:01 (+0000)  main  Effective FPS: 11.5
26.04.23 23:06:02 (+0000)  main  Effective FPS: 11.47
26.04.23 23:08:03 (+0000)  main  Effective FPS: 11.43

GOTO-GOSUB, 2023-04-27:
I get this if my display is on outside of the active hours (screenBlankHours variable) that I have set or if the trains are on strike again. I've also seen it on stations where the trains are very infrequent such as the Eurostar where the departure data does not appear to be provided all through the day. Assuming your choices of station have trains running at that time of night, may I suggest checking that you aren't running it outside of your active hours setting ?

ISC82, 2023-04-27:
Thanks for this. I checked those variables and they were correct. However, you made me think about some of the other variables and I then realised I had set the destination station. I cleared that and the departures started appearing again. So it seems that trains were being diverted to a different station in the past 48 hours. 

So lesson learned, check and double-check those variables first. Would have saved me a lot of wasted troubleshooting last night.

GOTO-GOSUB, 2023-04-28:
Cool, it's good to know you've fixed it. Thanks for the update as well since it might help someone else in future.

chrisys, 2023-05-08:
Thanks for helping here @GOTO-GOSUB ! Just to clarify, the reason that it does this is because that's what the signs say in the middle of the night when there are no trains (i.e. out of hours, or your filters have been set as in this case).

At least, this is what they used to look like last time I was waiting at a station in the middle of the night 😅