chrisys/train-departure-display #94: KITS AVAILABLE! 😮

chrisys, 2023-06-23:
Hey all. I have produced a handful of kits which I want to make available to help out those that don't have a 3D printer or easy access to get the display. Please don't expect perfect commercial 3D printing service quality but rather that of someone just trying to help out, so the parts might need some finishing work and will have minor blemishes. 

[image]

**They'll include:**
* Front housing
* Rear cover
* Desk mount option
* Hanging mount option
* 2 display clips
* Yellow or White OLED display (with the headers removed from the board so it will fit in the case)
* Screws
* Pre-cut USB power cable
* Handful of small wires in the right colours to match the documentation

**You need to provide your own Raspberry Pi Zero 2 W and SD card** - as much as I'd like to include these I can't get hold of them easily either! Assembly will require some basic electronics hand tools as per the instructions [in the documentation](https://github.com/chrisys/train-departure-display/blob/main/docs/01-getting-started.md).

**As of 23rd June 2023 I have 5 kits available (2 yellow, 3 white)**
When these are gone I have no idea if there will be more. I think a fair price for this initial test batch is £50 + postage to wherever. If you want one email me on chris@thefnet.co.uk and we'll figure out the details.

mattanorak, 2023-07-27:
If you ever do the kit without the screen that would be great.  Wouldn't mind putting mine in a case like that.  I tried to make a plexiglass cover - doesn't look bad, but it's not 100%.

chrisys, 2023-08-21:
@mattanorak send me an email with what parts you need - if it's literally just the 3D prints I might be able to help out, but I've run out of screens, wires, and USB cables (I think)

ruudvddries, 2025-02-23:
This one should work, right? https://nl.aliexpress.com/item/33012440441.html

> Not exactly the right place to post [@chrisys](https://github.com/chrisys) but didn't want to open a new issue. The Aliexpress link to the screen example no longer works, if you can find another suitable that would be handy, thanks. edit: Whilst looking (seems hard to find anything in stock that looks right) I noticed 3.12" seems more common? Guess it could be scaled up a little...

pauljrowland, 2025-03-05:
Getting there - just printing a deeper version of the screen as I've already destroyed one display taking the header off (the metal pads dropped out leaving just a hole with nothing to solder to 🤣 🤣) so decided to leave them be on this attempt. 

Really cool little project

https://github.com/user-attachments/assets/20f89390-8be5-4161-987b-7d9abe5a639f

[image]

closed 2025-06-10 by chrisys