chrisys/train-departure-display #99: Add in delay reason

MatthewAscough, 2023-07-14:
The API can report a delay reason when a service is delayed - this could be added after the TOC/carriage information

CalamityJames, 2023-07-19:
This is something I was also considering. I've been hoping that a delay occurs when I'm getting on a train so I can see how it is displayed on a real display, for some reason it's quite hard to find examples of these displays in action on YouTube!

chrisys, 2023-07-21:
@CalamityJames things that totally identify you as an "enthusiast", no sane person would ever say: "I've been hoping that a delay occurs when I'm getting on a train" 😆 

Presumably it's ENUM('Leaves on the line','Cows on the line','Driver is at the wrong end of the line','Someone said it might snow next Thursday')

MatthewAscough, 2023-07-21:
(un)fortunately? there’s delays on southeastern today cos of a broken track so i’ll try to find one on my way into work. i believe though it shows it after the carriage information so would be “— a thameslink service which has 8 coaches. — this train has been delayed by urgent repairs to the railway” but the API returns the full “this train has been delayed…” message so you can just pop that straight in after the hyphens

CalamityJames, 2023-07-21:
See I've been studying a few of my local stations (also Southeastern) and it seems there is no defined way of displaying this information.

Some show the delay info on a second scrolling message after the calling points. I think some do it other ways.

In fact the whole layout tends to vary from station to station. I could spend months making different layouts for different stations 😂

**East Farleigh**
_Smaller display as it's a tiny station, also coach information is in the top right and alternates between that and "On time" etc):_
[image]

**West Malling**
_Doesn't show the "1st" for next train, but DOES show 2nd. Also coach information appears top right again_
[image]

**Tunbridge Wells**
_Coach information shown in second scrolling line_
[image]

chrisys, 2023-07-21:
I wonder if these have all been made by different suppliers under different contracts over time depending on when the stations have been refurbished/updated. As for layouts the one I'd really like to add is an Underground one - with the cool alternative font (and different API).

CalamityJames, 2023-07-27:
I got a delay today! 🥳 

https://github.com/chrisys/train-departure-display/assets/5637797/d6b2a38b-d90a-4db8-86b3-dca2774f5fa1

It does seem like it's added as a new line, which complicates things a bit as I've been trying in vain to do that "pop up scrolly thing" in more than one place already.

I need to do some more researching into how Luma works I think!