chrisys/train-departure-display #100: Debug screen

stephenhibbert, 2023-07-15:
Spotted this debug screen today at Greenwich station, could be cool to have something similar to display the IP address on boot.[image]

CalamityJames, 2023-07-21:
I've created a mockup of this which seems to work well.

I have no idea what some of the codes on this screen signify, but here's what I've replaced:
[image]

1 = Current station code set in config
2 = Platform (shows PlatAll if no platform set) (todo: have different platform based on screen)
3 = Current version of Train Departure Display
4 = Today's date (can't think of an easy way to get the build date of the software without adding a new file)
5 = IP address

The next decision is **when** to show this. Open to suggestions here, especially from @chrisys !

I wondered about replacing the splash screen if a debug mode is enabled, or maybe even having a debug flag that leaves this screen on constantly.

chrisys, 2023-07-21:
Wow @CalamityJames I was just trying to clear my plate so I could have a look at this ticket this afternoon! This is great!

Here are my thoughts:
* I like your idea of a debug flag to show it, maybe something like if the var is set to TRUE it's always on but if it's an integer we show it on startup for that number of seconds? That feels like it would handle a decent amount of what people might want to do with it.
* My preference would be to remove information that doesn't mean anything in our scenario, but retain the look and feel, so maybe something like...
* Display = "<departureStation>[-><destinationStation>](Plat<platform>) <refreshTime>s <operatingHours>" so we'd end up with something like 'PAD->HWV (PlatAll) 120s 8-22h'
* Script = VERSION (I think we can just stick with the version here, as you say, date is a bit hard to do for the source, and I think today's would be misleading). Another option would be to get the build date of the current release from the balenaCloud API.
* Address = device MAC address? UID = balenaCloud device UUID (first 7 chars from BALENA_DEVICE_UUID), IP=local IP

What do you think?

CalamityJames, 2023-07-21:
Something like this? 😁 
[image]
(After putting a MAC address in I don't think there's room for UID)

Also, here it is without some extra options just for completeness
[image]

chrisys, 2023-07-21:
@CalamityJames looks great to me!

PR #101 (2023-07-22): Debug screen support
Fixes #100 

Adds a debug screen either replace the splash screen or the train departure info (permanent mode).

Setting `debug` to `True` replaces the main screen permanently.

Setting `debug` to an integer (except `1`, because `1 == True`) will replace the NRE attribution screen with the debug info for `debug` seconds. (This option hides the clock as it is a static display with `time.sleep()`)

I've tested it on a 3B with no detrimental effects but not got my Pi Zero boxed up into a display to real-world test it yet, although I don't anticipate any performance issues.

I [... cut]
closed 2023-07-22 by chrisys