chrisys/train-departure-display #105: Major disruption at Paddington causing crash

jruys, 2023-07-26:
“Major disruption” at PAD today. That’s the station my board is showing and (I guess) somehow the software doesn't like what its getting from the API. 

[image]
[image]

jruys, 2023-07-26:
Update: 2 minutes later board started running again, all 3 entries showing “delayed” or “cancelled”.

CalamityJames, 2023-07-27:
I believe I have added a `try` block for this error, however as it doesn't happen that often there isn't much real world testing that can be done here!

I made a few changes to how the `drawBlankSignage` function works, allowing it to display that there has been an error parsing the data, if the newly added debug mode is enabled.
[image]

If debug isn't enabled, it falls back to the Out of Hours station name, as we can't fetch the actual station name as there has been an error getting data from the API.

I did originally show this error message at any time, but I would imagine real departure displays wouldn't be verbose unless there was a debugging mode enabled, so added an if for debug.

It also should log the error to the console too.

I did consider writing a routine to temporarily speed up the `refreshTime` if an error like this occurs, but that might be out of the scope of this bug so decided against it for now!

I haven't submitted this a pull request yet, but feel free to check out the changes on my branch: https://github.com/CalamityJames/train-departure-display/tree/handle-openldbws-error

Will wait for @chrisys input to submit a PR!

BeauAgst, 2023-07-27:
Can I suggest that if we go down this route we put it behind a config? I'd much prefer my display to not update so that there's a more graceful degredation

CalamityJames, 2023-07-27:
That's a fair point, as it stands I think that by the point the loop has got here the old departure data will have been unset, I will try to have a play around this evening to see if there's anything that can be done

CalamityJames, 2023-07-27:
@BeauAgst I have added a variable that gets written to each time there is good data which can be used in the event of API failure.

I left the original error handling in, in case the display gets started and there is no good data at all

https://github.com/CalamityJames/train-departure-display/commit/e73754a60ae23603fca44d9108c2e4fbfadc4231

chrisys, 2023-08-21:
I think I'm of the view here that if we have good data (even if outdated) we can show it, but like you say if we start the display and there's no good data at all it's good to show the error message.