chrisys/train-departure-display #116: Screen stuck on 'UK Train Departure Display'

BowerSteve, 2023-12-22:
Hi - would love to get this working, but my screen is stuck on the 'UK Train Departure Display' - have not got as far as showing any platform departures.

Not seeing any specific SOAP errors in the logs, just the 'channel is already in use' error that seems to be benign:
 main  Starting Train Departure Display v0.5.4
 main  /usr/src/python-packages/luma/core/interface/serial.py:203: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
 main    self._gpio.setup(pin, self._gpio.OUT)
 main  Effective FPS: 12.31
 main  Effective FPS: 15.97
 main  Effective FPS: 17.23
 main  Effective FPS: 17.84
 main  Effective FPS: 18.2
 main  Effective FPS: 18.41
 main  Effective FPS: 18.62

Have tried tweaking variables and stations, but never get past this screen. 

```
debug	false
departureStation	STP
destinationStation	SRT
dualScreen	false
firstDepartureBold	true
fpsTime	10
individualStationDepartureTime	false
operatingHours	6-23
outOfHoursName	London St Pancras
refreshTime	120
screen1Platform	
screen2Platform	
screenBlankHours	0-5
screenRotation	2
showDepartureNumbers	false
targetFPS	20
timeOffset	0
TZ	Europe/London
```

Also tried re-generating a new API key - that's not the issue, initially the new key got SOAP errors and then those stopped when the token became valid after 10 minutes.

Any ideas on how to get some more diagnostics to know where the problem is?

[image]

bonstio, 2024-03-16:
This turned out to be a faulty display! It seemed like the screen was able to display data initially sent but nothing thereafter. New screen fixed the issue for @BowerSteve.

chrisys, 2024-04-23:
That's a strange one, but thanks for sharing the update guys, and glad you got it working in the end!

closed 2024-04-23 by chrisys