chrisys/train-departure-display #118: Successful use of 5.5" OLED (documentation)

jruys, 2024-02-03:
Felt like using a larger display so ordered a 5.5" OLED from Ali. 

Happy to report it works once you remove resistor P8080 to put it in 4SPI mode. 

Connector is different from the smaller display but signals are clearly marked. 

[image]
[image]

bonstio, 2024-02-03:
I'm inspired! Would you mind sharing a close up of the wires please?

jruys, 2024-02-04:
Let me post the Raspberry Pi pins to connect. Pi pin locations are listed in https://github.com/chrisys/train-departure-display/blob/main/docs/02-connecting-the-display-to-the-pi.md

Display - Raspberry

GND - 6
VCC - 1
SCL - 23
SDA - 19
RST - 22
DC - 18
CS - 24

Don’t forget to remove the resistor at P8080 otherwise display will not work.

closed 2024-02-04 by jruys

bonstio, 2024-02-05:
Awesome. Thanks!