chrisys/train-departure-display #124: Pretty sure this is NOT a code issue...

ScottTheHippy, 2024-05-14:
but damn.  After months of reliable enjoyment of my train departure display, I've entered into a peculiar realm where I'm killing Zero 2 W boards as fast I can order them.  I've tried different power- it all tests per spec.  Might the OLED be doing something toxic?  Has anyone seen this kind of weirdness?   

I expect not- so point, laugh, and close the issue if you wish.  I might be losing my marbles.   Cheers!

chrisys, 2024-10-31:
@ScottTheHippy I haven't heard of this in relation to this project but I have myself killed Pis in the past by accidentally shorting the power supply pins on the 40 pin header. I'll leave this open for now in case anyone else can chime in but I'd check for shorts on the power lines going to the display (whilst it's not connected to the Pi). For what it's worth I've got a couple of units that are both very old now -- from the very start of this project, and they're still going strong, so no long-term issues to report. Not even screen burn in as some were suggesting!

chrisys, 2025-06-10:
Will close for now (cleaning up issues) but feel free to reopen if anyone has any further experiences or input to share.