chrisys/train-departure-display #128: Display not functioning at all

novpaul, 2024-12-07:
Chris. first of all I wanted to say thank you for all of your work on this. This is my first Raspberry Pi project and I was inspired to do it by your work, so thank you. As a result I am a complete noob to Raspberry Pi, I hope this question isn't too tedious for you as it may be very basic.....  I have followed the instructions to the tee and believe I have the screen connected correctly, I have also tried moving R6 to R5. I am getting nothing from my screen, not any type of indication that it is functioning. The Pi is on (green light) and hopefully you can see from the attached that it is connected in Balena Cloud and operational. 

Is there something I am missing? Should I at least be getting some signs of life from the screen? I've no reason to think it's broken  as it's brand new. Is there a simple test I can do to check if the screen is working? Any advice would be super appreciated, really want to make this work. 

[image] 
[image]
[image]

novpaul, 2024-12-07:
[image]

novpaul, 2024-12-07:
[image]

bonstio, 2024-12-08:
Check your wires. Then check them again. Then re-do your soldering. Then try a new screen. I've built 10 of these beauties. One of the screens was either defective or I managed to trash it whilst soldering.

Good luck.

jruys, 2024-12-08:
First of all - what a great problem description! Not simply “it doesn’t
work” but stating what you already tried and clear pictures - bravo!

Can you get us a picture or double check the R5…R8 area? That got me a few
times. Those resistors are 0 ohm, so a blob of solder to bridge R5 and R8
should suffice. R6 and R7 should be open.

In the Balena console there’s a “logs” section under your device summary.
Anything interesting there when you (re)start the service?

Some other config things that caused my screen to go blank:

- “Dualscreen” was inadvertenly set to “true” and somehow that broke my
setup. That was a while ago, think that’s not an issue anymore.
- Time zone on my config was wrong which made the time fall within
“screenblankhours” which is when the screen is off! Mine is set to “22-7”
and TZ “Europe/London”.

Good luck!

JW

novpaul, 2024-12-08:
> Check your wires. Then check them again. Then re-do your soldering. Then try a new screen. I've built 10 of these beauties. One of the screens was either defective or I managed to trash it whilst soldering.
> 
> Good luck.

Thanks so much for the reply, the board came pre soldered with pins so I (hope) it's not a soldering issue. Is there a way that I can know if the screen is working, I'm a complete beginner so, for example, once the power is attached and the wiring is correct should I see some sign of life from the board even if it is fuzz or just lit up etc? Or can there be a software configuration problem i.e the screen will be black until the raspberry pi is configured correctly?

Someone mentioned to me that sometimes the raspberry pi is preconfigured to the hdmi port and so that needs to be changed? Or should it be the case if I followed Chris blog step by step out of the box that the display should work?

You're help is really appreciated.
P

novpaul, 2024-12-08:
> First of all - what a great problem description! Not simply “it doesn’t
> work” but stating what you already tried and clear pictures - bravo!
>
> Can you get us a picture or double check the R5…R8 area? [...]

Thank you also!

Do these pictures help?

[image]

[image]

[image]

[image]

novpaul, 2024-12-08:
This is what I'm getting from the log below. 

Just as an fyi I really am a beginner so even the most basic terms like leaving something "open" etc might have to be explained in laymans terms! Or even where the solder is supposed to go. 

I keep getting "key error soap Envelope" in the log if that means anything.  Thanks so much 

[image]

[image]

jruys, 2024-12-08:
I suspect there's something funky in your config - this is what mine shows on startup:

  Starting service 'main sha256:c41086cab1b98803fd20545d360830f14832dabadfb1dd87b701a714a3157097'
  Started service 'main sha256:c41086cab1b98803fd20545d360830f14832dabadfb1dd87b701a714a3157097'
  Releasing update locks
   main  Starting Train Departure Display v0.5.4
   main  /usr/src/python-packages/luma/core/interface/serial.py:206: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
   main    self._gpio.setup(pin, self._gpio.OUT)
   main  Effective FPS: 18.06
   main  Effective FPS: 19.0
   main  Effective FPS: 19.31

Yours seems to be unhappy about something in the config variables - it exits before the screen gets activated which would explain the blank screen.

Double check the "configuration" section? https://github.com/chrisys/train-departure-display/blob/main/docs/04-configuration.md

Re the screen - the interesting bit is what Chris shows in his manual [here](https://github.com/chrisys/train-departure-display/blob/main/docs/images/display-config-jumper.jpg) - the bits marked with R5 and R8 should have a R0 resistor in there, but a short piece of wire or blob of solder should do equally well. The bits marked R6 and R7 should have nothing between them (hence "open").

jruys, 2024-12-08:
...Probably not relevant but I'm confused - your log shows software version v5.5, I'm running v5.4 which as far as I know is the latest.

You did deploy with the big "deploy with balena" from the "installation" section of https://github.com/chrisys/train-departure-display/blob/main/docs/01-getting-started.md

Anyone else reading along: is there a 5.5 version?

metacurb, 2024-12-08:
Without being able to see the entire error, I would guess that your API details are invalid. The code is trying to reference fields in the API response that don't exist. Can you check your `apiKey` is set correctly, with no spaces and is in the right format?

metacurb, 2024-12-08:
You'll want to follow the final step in the docs - https://github.com/chrisys/train-departure-display/blob/main/docs/04-configuration.md

jruys, 2024-12-08:
> FYI @jruys , 0.5.5 is the latest version - https://github.com/chrisys/train-departure-display/blob/main/VERSION

Thanks @metacurb! Re-deployed and I’m now running v0.5.5. Looks like Balena still thinks it’s 0.5.4 though as it built it as a v0.5.4+rev1 version, probably something in a config file. Will check and file an issue if it’s not at my end.

jruys, 2024-12-09:
@novpaul The API key lives here in the Balena console:

[image]

novpaul, 2024-12-10:
Thanks all for the input-  will try the API key and see where that gets me Thursday/Friday

chrisys, 2024-12-13:
Just wanted to say thanks to everyone here for contributing and working to figure out what's going on!

@novpaul I would also share here the other values you've used in addition to the API key, maybe we can reproduce that way.

novpaul, 2024-12-23:
Hi all wanted to report back. I inputted the API key and no cigar unfortunately. I am now going to start from scratch again in the instructions and quadruple check pin out (I dont think this was incorrect) I have attached some screenshots if that helps from Balina Cloud... any further clues whilst working this out would be helpful as we crash into the Christmas period :-) 
[image]
[image]

novpaul, 2024-12-23:
[train-departure-display_calm-feather-2024-12-23t17_38_26+00_00.txt](https://github.com/user-attachments/files/18231926/train-departure-display_calm-feather-2024-12-23t17_38_26%2B00_00.txt)

Adding the log if that is helpful as well

novpaul, 2024-12-23:
Update: So I have redone the pin layout and it's definitely correct. I'm at a bit of a loss now. API key is correct. I'm getting the log below. Is there any way I can test the display to ensure the part is working, or is there more work needed (based on the log) to exhaust all avenues before blaming/investigating the screen itself for defect? Thanks - P

[image]

jruys, 2024-12-23:
Log looks promising, I would expect a boot screen with IP address etc show
up as a minimum but no luck. Back to hardware checking.

Another potential suspect could be the 4SPI hardware display setting.
Double check the section under "configuring the display" at the link below?
I know from experience if that's done wrong display also stays dark.

https://github.com/chrisys/train-departure-display/blob/main/docs/02-connecting-the-display-to-the-pi.md

Alternatively, send us a picture of that part of your display?

novpaul, 2024-12-24:
@jruys thanks again for taking the time to comment. I've attached the picture. Is the resistor the thing with three spots positioned in R6? Would I use a soldering iron to heat up the solder and remove it completely and then put a solder blob between R5 and R6, joining them together?

Hopefully this is the fix :)

novpaul, 2024-12-24:
@jruys thanks again for taking the time to comment. I've attached the picture. Is the resistor the thing with three spots positioned in R6? Would I use a soldering iron to heat up the solder and remove it completely and then put a solder blob between R5 and R6, joining them together?

Hopefully this is the fix :)

[image]

jruys, 2024-12-24:
Yes - Getting close...

The R5 area should look like R8 - a solder blob or miniscule piece of wire between the two pins (or a 000 resistor but don't bother - same thing).
The R6 area should be clear, just like R7 looks now.

novpaul, 2024-12-24:
Amazing all sorted thanks so much, what a lovely Christmas gift!! Thanks for everyone's help - super stars 

[image]

> Yes - Getting close...
> 
> The R5 area should look like R8 - a solder blob or miniscule piece of wire between the two pins (or a 000 resistor but don't bother - same thing).
> The R6 area should be clear, just like R7 looks now.

chrisys, 2025-08-08:
Thanks @jruys for helping out here - what a guy 🤩 Congrats on getting the project running @novpaul

closed 2025-08-08 by chrisys