chrisys/train-departure-display #130: CAD models?

jimmykl, 2025-01-09:
Any chance that you could provide CAD model files for the case components?

I'm looking to tweak the desk mount with a channel/hole for the USB cable and some way to add some weight (possibly a magnet)

MrDeformat, 2025-01-09:
Hey Jimmy, you should be able to use the files provided in certain software, I used tinkercad to do some rudimentary changes to my case.

chrisys, 2025-01-18:
@jimmykl as mentioned above you should be able to do this with the STL files. Is there anything else I can provide that would make it easier? If you make improvements, it'd be great to get them integrated back into the main repo - especially the weight/magnets idea! Mine is always falling over as well.

chrisys, 2025-06-10:
Will close this for now as I believe the STL files should be suitable - if not please reopen and let me know!

closed 2025-06-10 by chrisys