chrisys/train-departure-display #131: Can't get screen to work (using new 3.1" screen)

MrDeformat, 2025-01-09:
I successfully built the project last year with a 2.80" display (thanks Chris!) unfortunately it was above the kettle and it stopped working, so I ordered three sets of Raspberry Pi Zero and chose the slightly larger 3.12" displays now available.

- Have moved the 0ohm resister
- flashing with original code and I am getting a solid fps on Balena but blank screen
- have wired with simple push in wires

I'm out of ideas as to whats wrong is it my shonky wiring?

Tried flashing with a fresh install, but it gets stuck on: Creating network 'default'

[image]

[image]

[image]
[image]

chrisys, 2025-01-18:
Hey @MrDeformat - at a glance I'd say your solder connections need attention. Both the solder blob and the connections to the Pi look like they have the potential to be causing issues - I'm not saying that's definitely the case but you want to get the solder to 'flow' and fill the hole. It looks like the temperature on your iron was too low.

That being said I'm also unsure about these new displays - I'm assuming there are others that have had success with them, and I don't see why they wouldn't work if the controller is the same, but I don't have first hand experience.

91880, 2025-01-23:
Hi @MrDeformat !

Just came across the thread and I am in the same boat as you. I have exactly the same screen. I have doubled checked the pin spec and they are the same as @chrisys original guide. I have the jumpers correct to enable 4SPI. I've tripled checked my connections and they are all good with along with continuity on a meter.

I'm new to Balena and Raspberry Pi other than an internet radio project a few years ago, which is why I have a pi zero w which I am now using for this. 

Not sure what the log on my screenshot means as you have green text. 

No idea where to go next.

[image]

MrDeformat, 2025-01-27:
Hey @91880 looks like your board is stuck on supervisor starting, it's not loading up and sending any signal to the screen yet, I believe thats Balena's supervisor, the code that manages it, you may want to test previous supervisor versions, also double check your device configuration, and api key for any errors, you should be getting an updated fps rate in the logs, like my screenshot.

Let us know if you get it working, I'm certain its not the screen thats the issue, they are just plug and play and exactly the same layout as the last boards

91880, 2025-01-27:
Thanks for this @MrDeformat 

Will have a play and see if I can get any further forward. Will update on progress when I can.

91880, 2025-01-30:
Well this is proving challenging. I've tried three different SD Cards. All decent spec and wokring fine. I've tried flashing the image with both Etcher and with USBImager. I can see it in Balena Cloud each time but no progress on Supervisor State.

MrDeformat, 2025-01-30:
Thats frustrating, have you tried rolling back and using an earlier version, of both Balena, and the Train Time display software,

91880, 2025-01-30:
I can see I can select earlier OS versions, not sure which one I should go for? In terms of earlier version of the train display software, I just use the deploy to balena button. I don't know how to do it another way but I can learn!

MrDeformat, 2025-01-30:
Logged into my Balena and I think Supervisor version:14.9.4

Is the one that works for me

I will sit down ain a few weeks a properly look at the issue