chrisys/train-departure-display #135: Fixed but for info - unable to connect to WIFI / Balena Cloud

TellinStories, 2025-05-31:
Just in case anyone else has the same problem, I couldn't get this to work - it just wouldn't show as a device in Balena Cloud.  I worked out that it wasn't connecting to my router.  Firstly I made sure my router was using 2.4Ghz as the Pi Zero can't connect to 5 Ghz, however it still didn't work.  I eventually found out that it is because my router SSID contains a special character - I was able to go into the files on the SD card and manually enter my SSID and this fixed it.

Anyway, working now and love it!  Thanks!

chrisys, 2025-06-05:
Hey @TellinStories thanks for highlighting this and glad you enjoyed the project. I'll incorporate a note into the docs when I get a minute.

chrisys, 2025-06-13:
@TellinStories can you confirm what the special character was so I can attempt to reproduce this please?

TellinStories, 2025-06-13:
It is “&”Sent from my iPhone

chrisys, 2025-06-13:
@TellinStories thanks - and did you edit the `balena-wifi-01` file in system-connections? What change did you make?

chrisys, 2025-06-20:
Thanks for the extra detail @TellinStories! I have tried to replicate this many times now but have been unable to which is really strange. I can't what could be different on my end, so I'll leave this open for now in case someone else encounters the same problem.